MrBlinky/Arduboy-homemade-package #2: Linker error

sgoetze, 2018-11-12:
From time to time I come back to Arduboy to compile a new Arduboy game for my grandson who is visiting us - now with Arduino1.8.7. I have a self made Arduboy with 1.3 SH1106 OLED .

When compiling some new games, here for example tower defense, I got an error using the Arduboy optimized core. With standard core compilation and linking works but it gives less memory.

Here is the log. Possibly you can find out what has changed that it goes wrong now. Thank you in advance.

Arduino: 1.8.7 (Windows 10), Board: "Homemade Arduboy, SparkFun Pro Micro 5V - Alternate wiring, Cathy3K, Arduboy optimized core, SH1106"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\sg\AppData\Local\Arduino15\packages -hardware C:\Users\sg\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\sg\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\sg\Documents\Arduino\libraries -fqbn=arduboy-homemade:avr:arduboy-homemade:based_on=promicro_alt,boot=cathy3k,core=arduboy-core,display=sh1106 -ide-version=10807 -build-path C:\Users\sg\AppData\Local\Temp\arduino_build_827983 -warnings=none -build-cache C:\Users\sg\AppData\Local\Temp\arduino_cache_295247 -prefs=build.warn_data_percentage=75 -verbose C:\Users\sg\Documents\Arduino\Arduboy\microtd\microtd.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\sg\AppData\Local\Arduino15\packages -hardware C:\Users\sg\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\sg\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\sg\Documents\Arduino\libraries -fqbn=arduboy-homemade:avr:arduboy-homemade:based_on=promicro_alt,boot=cathy3k,core=arduboy-core,display=sh1106 -ide-version=10807 -build-path C:\Users\sg\AppData\Local\Temp\arduino_build_827983 -warnings=none -build-cache C:\Users\sg\AppData\Local\Temp\arduino_cache_295247 -prefs=build.warn_data_percentage=75 -verbose C:\Users\sg\Documents\Arduino\Arduboy\microtd\microtd.ino
Using board 'arduboy-homemade' from platform in folder: C:\Users\sg\AppData\Local\Arduino15\packages\arduboy-homemade\hardware\avr\1.2.5
Using core 'arduboy' from platform in folder: C:\Users\sg\AppData\Local\Arduino15\packages\arduboy-homemade\hardware\avr\1.2.5
Detecting libraries used...
"C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PROMICRO -DAB_ALTERNATE_WIRING -DARDUINO_ARCH_AVR -DARDUBOY_10 -DOLED_SH1106 -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"ABA1106\"" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\cores\\arduboy" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\variants\\arduboy-promicro-alt" "C:\\Users\\sg\\AppData\\Local\\Temp\\arduino_build_827983\\sketch\\microtd.ino.cpp" -o nul
"C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PROMICRO -DAB_ALTERNATE_WIRING -DARDUINO_ARCH_AVR -DARDUBOY_10 -DOLED_SH1106 -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"ABA1106\"" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\cores\\arduboy" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\variants\\arduboy-promicro-alt" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\Arduboy2\\src" "C:\\Users\\sg\\AppData\\Local\\Temp\\arduino_build_827983\\sketch\\microtd.ino.cpp" -o nul
"C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PROMICRO -DAB_ALTERNATE_WIRING -DARDUINO_ARCH_AVR -DARDUBOY_10 -DOLED_SH1106 -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"ABA1106\"" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\cores\\arduboy" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\variants\\arduboy-promicro-alt" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\Arduboy2\\src" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\EEPROM\\src" "C:\\Users\\sg\\AppData\\Local\\Temp\\arduino_build_827983\\sketch\\microtd.ino.cpp" -o nul
"C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PROMICRO -DAB_ALTERNATE_WIRING -DARDUINO_ARCH_AVR -DARDUBOY_10 -DOLED_SH1106 -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"ABA1106\"" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\cores\\arduboy" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\variants\\arduboy-promicro-alt" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\Arduboy2\\src" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\EEPROM\\src" "C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\Arduboy2\\src\\Arduboy2.cpp" -o nul
"C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PROMICRO -DAB_ALTERNATE_WIRING -DARDUINO_ARCH_AVR -DARDUBOY_10 -DOLED_SH1106 -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"ABA1106\"" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\cores\\arduboy" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\variants\\arduboy-promicro-alt" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\Arduboy2\\src" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\EEPROM\\src" "C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\Arduboy2\\src\\Arduboy2Audio.cpp" -o nul
"C:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_PROMICRO -DAB_ALTERNATE_WIRING -DARDUINO_ARCH_AVR -DARDUBOY_10 -DOLED_SH1106 -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"ABA1106\"" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\cores\\arduboy" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\variants\\arduboy-promicro-alt" "-IC:\\Users\\sg\\AppData\\Local\\Arduino15\\packages\\arduboy-homemade\\hardware\\avr\\1.2.5\\libraries\\Arduboy2\\s [... cut ...]
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.

cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.

Bibliothek Arduboy2 in Version 5.1.0 im Ordner: C:\Users\sg\AppData\Local\Arduino15\packages\arduboy-homemade\hardware\avr\1.2.5\libraries\Arduboy2  wird verwendet
Bibliothek EEPROM in Version 2.0 im Ordner: C:\Users\sg\AppData\Local\Arduino15\packages\arduboy-homemade\hardware\avr\1.2.5\libraries\EEPROM  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Homemade Arduboy.

MrBlinky, 2018-11-12:
Thanks for the report. Unfortunately I can not reproduce your error. I downloaded microtd and compiled with Arduino 1.8.7 'Homemade Arduboy' 'Sparkfun Pro Micro 5V alternate wiring' 'Arduboy optimized core' 'SH1106' successfully.

I recommend to go to board manager, search for home then downgrade the version and upgrade right away again to 1.2.5 so the package will be reinstalled. Close Arduino IDE and try again. If that still doesn't help reinstall Arduino IDE or try testing with Portable Arduino IDE version

sgoetze, 2018-11-13:
Now I have tested the Arduboy-homemade-package on a Debian VM first with Arduino version 1.8.4 (and the same Arduboy parameters as above) and finally with version 1.8.7.

Both version compiled microtd successfully. So I have no idea what the problem is with my Windows 10 Arduino installation. I upgraded to Arduino version 1.8.7 just before I wrote about the linker error and I even tried some other sketches with the same error. I did not reinstall the complete Arduino IDE because I use and need some board plugins. I'm not shure if there is an interaction between the plugins. I even have installed Atmel Studio 7.0 (upgraded to the newest version some weeks ago) and possibly there is some influence to the tool chain for AVR MCUs. 

But this is not your problem and with a VM I have a way to come around the linker problem. So it's no issue of your package. I only post this as a comment but I think you can close this issue.

MrBlinky, 2018-11-13:
You can use the Arduino portable version side by side with the installer (or multiple portable versions). That's how I test compatibility between different IDE versions. You just unzip the download and create a portable folder inside the Arduino folder and run the executable. Much easier then setting up a VM.

I'll close this issue.

closed 2018-11-13 by MrBlinky