MrBlinky/Arduboy-homemade-package #5: Bring out A5 on expansion connector

MLXXXp, 2020-04-30:
Suggestion:
Pin 8 on the expansion connector is currently assigned as:
`Not connected (reserved for future use)`

I suggest specifying it as Arduino Pin A5. This pin is not used on the Arduboy but is brought out as a pad on the back of the circuit board.

Having an analogue input would allow potentiometer based paddles or other variable voltage devices to be more easily interfaced. Having only one analogue input might be a bit limiting but it could be expanded using an analogue multiplexor IC, controlled by I²C, SPI or digital.

Of course, A5 could also be used for general purpose digital I/O or as another CS for SPI.

MrBlinky, 2020-05-01:
I wasn't sure what to hook to the unused pin yet. But A5 makes the most sense. (easy accessible and an analog input) I'll adapt the documentation.

closed 2020-05-01 by MrBlinky

MLXXXp, 2020-05-01:
You may wish to note in the documentation that A5 isn't available as a brought out pin on a standard Pro Micro.

MrBlinky, 2020-05-01:
Good point. Added.