MrBlinky/Arduboy-homemade-package #6: Probelm with Arduino IDE 1.8.13 (Windows 10)

venice1200, 2020-06-24:
Hi,   
I am using the Windows ZIP Versions of the Arduino IDE and tried the actual v1.8.13 
using the configuration from the previous (working) Versions 1.8.12 and 1.8.10.   
If I start v1.8.13 with an previously used "normal" AVR Board and choose now one of the Arduboy Boards   
I get some Java exeptions shown and the IDE 1.8.13 is no longer able to start if stopped.   
v1.8.12 is still working.

Is the problem related to the Arduino IDE 1.8.13, to my Windows or to the Arduboy-homemade-package?

Thanks for your Package, the Bootloader and all the other cool Arduboy stuff.

Cheers
venice1200

[Arduboy-homemade-package_Arduino_IDE_1.8.13_exeptions.txt](https://github.com/MrBlinky/Arduboy-homemade-package/files/4827812/Arduboy-homemade-package_Arduino_IDE_1.8.13_exeptions.txt)

MrBlinky, 2020-06-24:
Thanks for the report. I just downloaded the new 1.8.13 IDE verison and can confirm the issue. I will look into it.

MrBlinky, 2020-06-24:
Looks like the new feature of only showing compatible programmers crashed on a null pointer because there is no list of compatible programmers.  When that happens DON'T go to the board manager it will crash and cause the IDE to crashing on starting the next time. I think adding at least one compatible programmer will fix the issue

MrBlinky, 2020-06-25:
found the problem. Arduino IDE 1.8.3 doesn't like it when there is no default .build.core= line for a board in boards.txt. 

adding the following lines to the board.txt file will fix the problem.

```
arduboy-homemade.build.core=arduino:arduino
arduboy.build.core=arduino.arduino
arduboy-devkit.build.core=arduino.arduino

```
Will add a fix tomorrow

venice1200, 2020-06-25:
Hi,
I can confirm the fix with the additional lines.
Many Thanks for your help.

venice1200, 2020-06-26:
Just as an request from my side;

I think it makes sense to rename in boards.txt the line
arduboy-homemade.name=Homemade Arduboy
in 
arduboy-homemade.name=Arduboy Homemade
just from an cosmetic point of view.

If you use the cli version of arduino the arduboy version/boards are all listed together after this change 
as the list is alphabetical.

MrBlinky, 2020-06-26:
Thanks for your suggestion. But i'll keep it Homemade Arduboy. You're free to rename it in your version ofcourse.

MrBlinky, 2020-06-28:
The issue is fixed now in 1.2.8 package

closed 2020-06-28 by MrBlinky