MrBlinky/Arduboy-homemade-package #7: [Question] What options do I select for a common Pro Micro?

menehune23, 2021-01-18:
In your table on the readme, you mention "Pro Micro", but in the IDE menu, it says SparkFun Pro Micro. Is there a difference? Common Pro Micros from ebay, etc contain a Leonardo bootloader, but SparkFun's is their own, I believe.

Side question: With the Pro Micro standard wiring, I've also gotten this to work with the standard Arduboy libraries. I'm trying to figure out if I should use your package or the standard libraries (you said you cannot have both installed [do you have docs that mention the order of precedence? I thought package libraries would already take precedence]). Does your package do anything special with the speaker and LED wires that I wouldn't get with the standard libraries? (I'm using a SSD1306/1309 screen, so I'm as close to the original Arduboy as I can get).

MrBlinky, 2021-01-18:
The difference between the Sparkfun Pro Micro and the (clone) Pro Micro is in the bootloader but that is not relevant for the homemade package. I'll add parentheses or remove the Sparkfun bit to prevent the confusion.

The standard wiring is the Same as Arduboy (Leonardo) wiring but since the display reset, green LED and 2nd speaker pin are not broken out on a Pro Micro those features can't be used.

Manually installed libraries (in your sketchbook) take precedence over the ones includes by the board package.
So Installing the standard Arduboy library will render the Homemade package settings useless.

If you use a Pro Micro with standard wiring and an SSD1306 display you don't need to use the package if you.

* Don't care about green LED
* don't care about 2nd speaker pin.
* Not going to add a flash chip
* use a SSD1306 OLED display
* not make use of the Cathy3K bootloader and use the extra 1K of flash
 
Note that uploading a sketch using the standard libraries the display will not look correct on a SSD1309 display (left and right halves of the display are flipped). 

If you do want to use both homemade package and original libraries. You can make (one or more) Portable Arduino IDE versions and install the Homemade package in one version and the original libraries in the other.

menehune23, 2021-01-18:
Thanks for the clarification! That's good to know about the SSD1309. I'll stick with your package then. If I wanted to use your package and program both an official and homemade arduboy, I can still select the standard "Arduboy" from the homemade section of the menu right?

Also: I thought the display reset is on the Pro Micro, at D6?

menehune23, 2021-01-18:
Also, does your package support a homemade arduboy WITHOUT the flash cart, but with the ability to use those pins for other things? I didn't see a "None" option in the bootloader menu. I'd want to make sure that whatever I select doesn't make those pins (basically the cart CS) unusable for other purposes (sensors, etc).

menehune23, 2021-01-18:
I'll close this as answered, but feel free to comment and answer any remaining questions :)

closed 2021-01-18 by menehune23

MrBlinky, 2021-01-18:
> I can still select the standard "Arduboy" from the homemade section of the menu right?
Yes. but you'll compile  with the modified libraries of the package.

> I thought the display reset is on the Pro Micro, at D6?
Yes for Arduboy and PRo Micro standard wiring. On Alternative wiring the pin is used as 2nd speaker pin.

> does your package support a homemade arduboy WITHOUT the flash cart,
Haven't thought about that. I'll take this into consideration.

menehune23, 2021-01-18:
Ah, I realized I was assuming the flash chip was optional. And technically it is if you don’t plan to use the cart CS for anything else, right?

In the case of wanting it to be available for general use, it would be nice to be able to tell your package not to try to do anything with it.

MrBlinky, 2021-01-19:
The flash chip is optional. But will become mandatory for Arduboy FX games. Anyway I can't imagine someone not wanting the flash with all the games playable on the go.