MrBlinky/Arduboy-homemade-package #8: [Feature Request] Non-LED option

menehune23, 2021-01-18:
It would be nice to have a menu option for wiring that does not include the RGB LED (the library could NO-OP the API for it). It would be nice to have these available for other purposes (i.e. for use in the extension header). Thanks and great work!

MrBlinky, 2021-01-18:
That's out of scope of the Homemade package. If those specific pins are required for other purposes, choose the standard wiring and Use an ItsyBitsy or Teensy2.0 (clone) Those have the pins missing on the PRo Micro broken out and more.

closed 2021-01-18 by MrBlinky

menehune23, 2021-01-18:
Sorry, I may not have been super clear. I’m thinking of the common case where folks use the standard wiring on a Pro Micro, but choose not to hook up the RGB LED. I get that there’s already one color channel missing, and would only want to ensure the library doesn’t try to do anything with the remaining two, in case folks want to use them for other purposes without colliding with the library.

It would be a pretty simple macro to NOP the LED code based on a flag. I may PR it if you’re willing to consider it in scope, as it would seem to follow the purpose of “making homemade Arduboys easier”. Thoughts?

MrBlinky, 2021-01-19:
The package is focused for making a homemade Arduboy. The standard wiring scheme is intended for the ease of uploading original hex files.  The alternative wiring to support most hardware features of Arduboy.
Using GPIO pins for other uses is not part of it.

If you wish to connect other hardware you can reconfigure the pin functions after Arduboy.begin/boot. If there would be be a conflict you can make an adapted version of bootpins.