MrBlinky/Arduboy-homemade-package #9: [Feature Request] Assert display CS in software only when in use

menehune23, 2021-01-18:
IIRC, the Arduboy2 library asserts the display CS on bootup, and this package has some wirings just ground it in hardware. If the arduboy library instead only asserted CS during a frame render, it could make SPI more easily available to other devices for extensibility. I've done this before (though I don't have the code any longer), and it worked out nicely, since there is no DMA or other async work happening during a frame render on the 32u4.

Note: this change would also probably be good to PR upstream to the Arduboy2 library.

MrBlinky, 2021-01-18:
THe OLED display and flash chip both share the same SPI bus. grounding the displays CS is not an option.

Because the offical library doesn't support it (and most likely will not be implemented too due to increased code size).  It's the others SPI device codes task to disable and enable the OLED chip select.

closed 2021-01-18 by MrBlinky

menehune23, 2021-01-18:
Ah that’s a nice idea too. I like that approach.

menehune23, 2021-01-18:
One downside though is that then every device has to be responsible for every other device, rather than just itself. For instance, it’s much easier to reason about the code and less error prone to just assume every device cleans up after itself by deasserting its CS when its work is done (assuming only synchronous code).

I get the code size argument though. I do wonder how much it would increase, as it just feels wrong for the display to hog the SPI bus.