MrBlinky/Arduboy-homemade-package #10: RGB LED strange behavior

menehune23, 2021-01-27:
Hi there, I tested out the RGB LED example on a homemade arduboy (alternate wiring), and the example turns on/off the LEDs properly for the "digital mode", but not for "analog mode". I tested this with an official arduboy and got the same result, so I believe the issue is with this package. I also tested with the official arduboy2 library's example on the official arduboy hardware and the example worked perfectly.

MrBlinky, 2021-02-06:
Thanks for reminding.

The homemade package uses an optimized version of the old Analog RGB mode because the new analog code in the official Arduboy2 library caused problems for a game. That game has been updated now so I'll also implement the new code

menehune23, 2021-07-12:
Hey there, just following up on this. Would be good to have working LEDs for homemade stuff :)

menehune23, 2021-11-15:
Hi, just bumping this again. Happy to also look into it if you can offer any pointers.

MrBlinky, 2021-11-26:
Thanks for the bump! will try to update the library soon.

MrBlinky, 2021-12-05:
Changed RGB LED code to match official Arduboy2 library

closed 2021-12-05 by MrBlinky