MrBlinky/Arduboy-homemade-package #11: Is there possibility to add support for screen with SSH1106 over i2c ?

GamerClassN7, 2021-02-03:
Hi,
Is there possibility to add support for screen with SSH1106 over i2c ?
thank you in advance for answer

GamerClassN7, 2021-02-03:
[image]

GamerClassN7, 2021-02-03:
IT IS MOST COMMON oled IN MY COUNTRY
LINK TO DATASHEED https://drive.google.com/file/d/19hxmIwuuTNxKzC-OcNMQJkltGV3B0DgB/view

MrBlinky, 2021-02-06:
I'll look into your request. I see you’ve also asked the same question in [Arduboy Community](https://community.arduboy.com/t/ssh1106-i2c/9583). Please continue communication there.

GamerClassN7, 2021-02-08:
@MrBlinky  i added required information to Arduboy Community. :)

MrBlinky, 2021-12-05:
Support for SH1106 I2C display added

closed 2021-12-05 by MrBlinky