MrBlinky/Arduboy-homemade-package #14: Question: Flash CS pin

menehune23, 2021-11-15:
This is more of a question than a request for any change. I'm trying to understand the use of the flash CS pin.

- Is the bootloader the only thing that uses the flash CS pin? Is it possible to use this as a GPIO within games if not using a flash chip?

- Is my understanding correct that to select a bootloader from the menu means you have that bootloader flashed to the Arduino chip already? Are these options irrelevant if you've got the standard Arduino bootloader installed?

- How does the bootloader know which flash CS pin to use? I see that there are two options for which pin to use, based on wiring. Is each bootloader compatible with each pin? Or is it such that Cathy3K uses the newer pin configuration (i.e. SDA as CS pin) while the other bootloader option uses the older pin configuration (D2 for CS)?

Thanks for the clarification!

MrBlinky, 2021-11-26:
The Flash CS pin is used to only select the Flash chip when required. Both the OLED display and flash chip  use the same SPI bus and FLASH CS and OLED CS are used to select one device at a time (In certain cases both devices can be selected at the same time though for some smart trickery but I wont go into details on that here).

Also you can't just ground FLASH CS like it can be done for the OLED display. Because the FLASH CS must be asserted and deasserted for a  flash command to executed or ended.

The Cathy3K  bootloader uses FLASH CS to detect a flash chip and read menu data and hex files from it.
Normal (Non Arduboy FX) sketches don't use FLASH CS so it's available as a GPIO pin. 

The bootloader setting in the Arduino IDE is only relevant for 2 things:

1) If you want to burn the bootloader through the Arduino IDE
2) If  you want to compile a sketch for an (DIY) Arduboy with Caththy3K bootloader that makes use of the 1K extra (PROGMEM) flash. So a sketch is not limited to 28K but 29K.

menehune23, 2021-11-27:
Thanks for the info. It's super helpful!

For [dotMG](https://github.com/menehune23/arduboy-dotmg), I'm using a [wiring scheme based on your Pro Micro "Alternate" wiring](https://github.com/menehune23/arduboy-dotmg/blob/main/hardware/system/schematic.pdf), except I've adjusted the display reset pin to allow SDA to be free for extensibility (including using a flash chip).

To free up the SPI bus for runtime add-ons, I've also adjusted the library code to only assert the display CS when needed ([diff here if curious](https://github.com/menehune23/arduboy-dotmg/commit/f1758e9036c78f26eb6328505f2e0944954ad56e)).

menehune23, 2021-11-27:
Side note: Although this means that any .hex files (like those used in the standard FX image) won't be natively compatible. I'm using the 2.42" OLED 1309 display, which has a way to patch .hex files, but I'll have problems with the display reset pin being switched to D0. Is this sort of patch difficult to do? Any advice on writing a hex patcher for a change like this? Do you do anything like this for your other wiring schemes? (I see that they don't all use the same display reset pins).

menehune23, 2021-12-04:
Closing as you've answered my original questions. I went and asked the side questions over on the [Arduboy forums](https://community.arduboy.com/t/dotmg-formerly-arduboy-cartridge-edition/9550/181?u=menehune23) instead.

closed 2021-12-04 by menehune23