MrBlinky/Arduboy-homemade-package #15: `BeepPin2` doesn't always work

menehune23, 2021-11-19:
I noticed that `BeepPin2` didn't seem to work with my homemade (Pro Micro-based) Arduboy. I then realized that the code in `Arduboy2Beep.cpp` doesn't have any adjustments based on wiring. I'm not sure if it will be possible to make this work for certain wiring scenarios, due to which timers are connected to which pins, but at the very least you might consider issuing a compiler error with a helpful message.

menehune23, 2021-11-19:
Turns out that for the alternate wiring at least, it's possible to make `BeepPin2` work by setting the following instead of `TCCR4A` (based on the ATmega32U4 data sheet):

```cpp
TCCR4C = bit(COM4D0);  // connect Arduino pin 6 (toggle mode)
```

Tried it out and it works! I have the [change over on my fork for dotMG](https://github.com/menehune23/arduboy-dotmg/commit/54a988f214f234bd03e67d6ffe2466094b158986), but my scenario is a little different from yours. Feel free to use it as a reference (not that you need it 😉).

I only have one wiring scheme to worry about. You might want to just omit compiling `BeepPin2` altogether for any wiring schemes that don't support a second speaker pin.

MrBlinky, 2021-11-26:
Thanks for reporting the issue. Arduboy2Beep.cpp was added at a later stage and I didn't adapt it for alternate wiring. Will have a look at your sugestion and I'll update the library.

MrBlinky, 2021-12-05:
Added support for alternate wiring to Arduboy2Beep.cpp

closed 2021-12-05 by MrBlinky