MrBlinky/Arduboy-homemade-package #19: There is no Programmer option under the Tools menu

zipeijun, 2023-06-05:
Arduino IDE 2.1.0 has no Programmer option under the Tools menu, Arduino IDE 1.8.19 behaves normally and is able to flash the bootloader.

**Arduino IDE 2.1.0**
[image]

**Arduino 1.8.19**
[image]

MrBlinky, 2023-06-05:
Hi! this is a known issue. The Arduino 2.x IDE has different behaviour. I found out the new IDE needs it's own copy of programmers.txt file in the boards package folder (location of boards.txt) This will be fixed in the next update.

For now you can copy the programmers.txt file from .../hardware/arduino/avr to ...packages/arduboy-homemade/hardware/avr/1.3.8

MrBlinky, 2023-06-05:
You also need to remove the cached files so the IDE 'sees' the changes.  Removing the folder .../AppData/Roaming/arduino-ide will be the easiest way to do this.

zipeijun, 2023-06-05:
Thank you for your reply, I have solved this problem with Arduino 1.x
I am raising this issue to help others with the same problem, for which I have not found a solution online.
Hopefully, can add a note to REDEME to help others.

(I am not a native English speaker, so I hope I can express my meaning clearly)

closed 2023-06-07 by zipeijun